Hackuarium/bioreactor #24: Weight still gives some wrong values

lpatiny, 2020-06-14:
In order to have reliable weight value we could try to:
- change the priority of the thread
- calculate the median of 10 values
- reduce the allowed difference between acquisition. Should be less than 2% probably

andrescvargasr, 2022-05-03:
There is a hardware issue with the weight, everytime that the protocol restart, the EMPTY and MAX are in other place in the flask (test with no change and avoid touch the bioreactor). Try to check the HX711 datasheet for more information.

lpatiny, 2022-05-04:
@andrescvargasr did you try to look at the raw value ? Can you should example of the raw values after each restart in this issue ?

andrescvargasr, 2022-05-04:
Some help:

https://hackaday.io/project/1741-honeybee-hive-monitoring/log/9780-modifying-the-hx711-breakout-board-for-33v-operation

andrescvargasr, 2022-05-04:
We saw the raw values and we obtain different values in static test for the wieght, apparently we need to change the values for the resistors in the board (move compatibility for 5V to 3.3 V).

andrescvargasr, 2022-05-05:
For 3.3 V power source in the  analog stage in the HX711 we need to replace the resistor R31 in the bioreactor v6.0.1 from 8,2 kOhm to **15 kOhm** to assure than V_AVDD is less than 3.3 V - 100 mV = 3.2 V. In this case, with R30 = 20 kOhm and R31 = 15 kOhm, we obtain: V_AVDD = 2.92 V. With this change we can assure the transistor is ON and it can send enough current to the load cells.

Equation:

V_AVDD = V_BG * (R30 + R31) / R31; V_BG = 1.25 V, R30 = 20 kOhm, R31 = 15 KOhm.