Hackuarium/bioreactor #27: bug: bioreactor stepper is being noisy

opatiny, 2020-06-16:
The cause might be an input current too big. Checkout the [datasheet](https://www.ti.com/lit/ds/symlink/drv8811.pdf?ts=1592288897741&ref_url=https%253A%252F%252Fwww.google.com%252F) to find if one of the resistors doesn't have the right value.

Screenshot of the resistors of the stepper:
[image]

opatiny, 2020-06-16:
After inspecting the datasheet, we could calculate the relationship between the max current per coil of the stepper, RSENSE and RLIM. We came to the conclusion:
 - RSENSE = 0.1Ω => RLIM = 82Ω
 - RSENSE = 0.2Ω => RLIM = 168Ω

This should suit a stepper with 0.1A per winding.

No other explanation that a problem with RLIM was found which would explain why the stepper is noisy and why it gets really hot.

opatiny, 2020-06-16:
Some of the noise can be caused by vibrations of the stepper motor -> we should add some rubber spacers around the screws holding it in place to remove some noise.

opatiny, 2020-06-17:
In this comment, I describe the formulas to calculate RLIM and VVREF depending on the other known parameters.

[image]

Description of the variables:
- VVREF: voltage on pin VREF. This value limits IFS.
- IFS: the amount of current per coil of the stepper the chip outputs.
- RLIM: resistor R47 on the image above. Changing this resistor allows us to modify VVREF.
- R: resistor R41 on the image above. R=330Ω
- RSENSE: resistor of the current sensor of the chip. RSENSE = 0.2Ω
- VCC: chip input voltage. VCC = 3.3V

From the datasheet, we have the following formula:
[image]

Additionally, after some basic equations solving, we obtained the following formula:
[image]

Which is equivalent to:
[image]

opatiny, 2020-06-17:
> After inspecting the datasheet, we could calculated the relationship between the max current per coil of the stepper IFS, RSENSE and RLIM. We came to the conclusion:
> 
> * RSENSE = 0.1Ω => RLIM = 82Ω
> * RSENSE = 0.2Ω => RLIM = 168Ω
> 
> This should suit a stepper with 0.1A per winding.

RLIM was changed twice to try these values. We observed that the torque of the stepper got really weak, to the point where the stepper wouldn't start spinning by itself. ( You had to push it a little bit to get it turning.)

This problem was solved when ramp in and ramp out were implemented (issue #23).

In the end, though, we put back a 330Ω resistor. After some testing, we realized that the stepper was not heating anymore.

closed 2020-06-17 by opatiny

lpatiny, 2020-09-26:
For RSENSE = 0.1Ω a value of 180Ω for RLIM seems good.